FRCDashboard/FRCDashboard #3: Usage with USB webcam

calebeby, 2016-07-29:
Hello,
I'm from team 1432 and we're looking into using your dashboard. So far it looks great, except I can't figure out how to use this with a USB webcam. I realize that I have to change `style.css` and the background image url, but I do not know what to change this to.

Thanks, Caleb

ErikBoesen, 2016-07-30:
Glad to hear you're interested in using my application!

This is one of the challenges I'm going through in making the dashboard more easy to mess with.

Since there's a variety of ways to hook up a camera, I'm not sure exactly what to tell you. **However**, I can tell you how our team does it. I think that should get you on the right track.

We use the MPEG Streamer library (I think [this one](https://github.com/jacksonliam/mjpg-streamer)). If you open the configuration page up while connected to your robot, there's a selection of ways to view your camera output. Find whichever one provides raw images. Get the src of the image. Put that into the stylesheet in the obvious place. It should work.

I realize what a terrible explanation that is, and I'm sorry, I just don't know that much about how the camera streamer system works yet. If it helps, I'm currently redeveloping this year's version of our team's UI, off which FRCDB is based, with the extremely simplified FRC Dashboard library. You can find it, and see how I did the camera for it, [here](https://github.com/frc1418/FRCDashboard).

If you already have a different type of dashboard with a streaming system, you can take a look at that and see how it works or where it's getting its source from.

If you want to give me a link to that, feel free to do so and I'll see if I can help.

If you need more help (with this or anything else), feel free to ask.

Best of luck and thanks again for using FRC Dashboard!

calebeby, 2016-07-30:
Last year we were using the java smartdashboard, but I have no idea how that works

ErikBoesen, 2016-07-30:
Does that have camera support? I'm pretty sure it doesn't, but there may be something I don't know.

calebeby, 2016-07-30:
Yeah, the camera stream is there.

calebeby, 2016-07-30:
I installed mpeg-streamer via https://github.com/robotpy/roborio-packages

calebeby, 2016-07-30:
Then I rebooted the roborio, and went to http://roborio-1432-frc.local:5800/ , but it didn't load

ErikBoesen, 2016-07-30:
Oh, yeah, sorry, that's the one I was looking for.

That IP might not be a global thing. I think it's something just our team uses.

ErikBoesen, 2016-07-30:
https://www.chiefdelphi.com/forums/showthread.php?t=101314

I'm looking at this.

Do you know how you can get the IP of your camera? Maybe ask your mechanics team. They should know.

calebeby, 2016-07-30:
https://www.chiefdelphi.com/forums/showthread.php?t=101314

they talk about configuring the camera with the tool that frc provided. Any idea what that tool is?

ErikBoesen, 2016-07-30:
Sorry, I forgot to push my code. Should be there now.

calebeby, 2016-07-30:
You are using a usb webcam, right?

ErikBoesen, 2016-07-30:
Hmm, I don't know what utility they're speaking of. Let me ask some of the others on my team.

And yes, we are.

calebeby, 2016-07-30:
10.14.32.2?action=stream does not load either. When I go back to robotics on Monday, I will check and see if I can find another ip for the robot. Thanks

calebeby, 2016-07-30:
whoops didn't mean to close this.

ErikBoesen, 2016-07-30:
That's just the IP of our camera. Probably not yours. Ask around your team and see if anyone knows the camera IP or how to get it.

Sorry I haven't been so helpful, please tell me if you have any ideas to improve the explanation in the `README` et al.

Also, if you ever need to manage input from multiple camera streams, I'm about to update the [MultiCamera addon](https://github.com/FRCDashboard/addon-MultiCamera). It's currently broken but should be fixed by tomorrow.

ErikBoesen, 2016-08-04:
Did you learn anything from the rest of your team?

calebeby, 2016-08-05:
No. I'm the only programmer, except for our programming mentor, who I might
get to talk to on Friday. I emailed @virtuald regarding this, and he gave
me a few suggestions. I emailed him back, but he hasn't responded yet.

Thanks for your help!

I will get back to this thread once I get a response from Dustin.

ErikBoesen, 2016-08-05:
Excellent, glad to hear you're getting somewhere with this issue.

calebeby, 2016-08-18:
OK. I figured this out with @virtuald. It turns out that `roborio-packages` was missing `libv4l` when it installed, but we got it worked out. Thanks!

closed 2016-08-18 by calebeby

ErikBoesen, 2016-08-18:
Excellent, glad to hear. Feel free to open another help wanted issue if you have any other questions.