FRCDashboard/FRCDashboard #8: Feature Request: Asset Compilation

calebeby, 2016-08-26:
I think it would be really helpful to let people use languages such as SASS, Stylus, CoffeeScript, and Jade, and have these compiled into HTML, CSS, and JavaScript.

I was able to get this working using [`electron-compile`](https://github.com/electron/electron-compile), but this is done on-the-fly, and I think that it would be better to use something like [`brunch`](http://brunch.io/).

I have not yet used brunch with electron, but we could probably start brunch using `child_process`, and then change the electron url to the public folder, which is generated by brunch.

To restate that, I think there are three options:

1. Leave it as it is, with no fancy assets :disappointed:
2. Use `electron-compile` to serve the assets, but some things don't work like imports
3. Use `brunch` to generate the assets into a folder

calebeby, 2016-08-26:
If you are interested, you can check out 1432's [test dashboard](https://github.com/team1432/test-dashboard), which uses `electron-compile`

ErikBoesen, 2016-08-26:
Okay, this sounds like a neat idea. I've never worked with this before, but I'd be interested to try.

Also, in the future, please fork my repositories instead of just copying them.

calebeby, 2016-08-26:
Ok. I'm working on a pull request with brunch, It'll probably be done in a couple days

calebeby, 2016-08-26:
Nvm got it done today. Find and replace ftw!

closed 2016-08-26 by calebeby