FRCDashboard/FRCDashboard #13: Load/include addons instead of making the user copy/paste

virtuald, 2016-09-09:
It seems a bit clunky to me to tell users to copy/paste HTML/JS/CSS from your addon repos. It'd be nicer if you could just copy the files verbatim to your custom dashboard, and then put some magic on the main page that says "insert addon foo here". 

I'm not really sure of the best way to do this, but I have to imagine there's a way. Maybe @amorygalili knows.

Another way to go about this: Frameworks like ember or Webpack have a custom packager that takes all the files and packages them together to run them. That's probably far more complex than what you need, but perhaps there's some good stuff to investigate there.

ErikBoesen, 2016-09-09:
I've been trying to think about how to do this for a while. The main thing stopping me is that most of the addons require adding to the `switch` statement in the `onValueChanged` function. Everything else would be mostly modular, except that. Any idea how I can get around this?

amorygalili, 2016-09-09:
I agree with Dustin. Maybe each addon should be contained in a single
folder and adding an addon would be as simple as dragging it into an
"addon" folder.

I havent been able to look over your code yet so I don't know what your
switch statement does. If you need to expand the switch statement every
time you add an addon maybe each addon could have a js file that handles
this part. You could use js to append <script> tags to the body which would
load the js files from each addon. Your program will have to know the paths
of these script files to load them which is tricky. You could add server
code which lets you get all the folders in the addon directory. Or maybe
you could have a config file with an array of the addon paths that the
developer would have to modify if they want to use the addon.

If you need to load files asynchronously  in a specific order I would also
look into requireJS. It's not a framework like emberJS and it's flexible,
so you don't have to learn a ton or change your project a lot to get it to
work.

virtuald, 2016-09-14:
Regarding the switch statement: store functions in a dictionary (pynetworktables2js has one borrowed from d3).

```js
var d = d3_map();

function dashboardRegister(key, fn) {
  d.put(key, fn);
}

function onValueChanged(key, value, isNew) {
  var fn = d.get(key);
  if (fn !== undefined) {
    fn(value, isNew);
  }
}
```

Then you can register them later on:

```js
function myOnValue(value) {
   $('#y0').css(...);
}

dashboardRegister('/path/to/key', myOnValue);

```

There's a lot of different variations on this. One that I would recommend is defining a FRCDashboard object, and putting that stuff there so that it's something like `FRCDashboard.register(key, fn)`.

ErikBoesen, 2016-09-14:
Related: https://www.chiefdelphi.com/forums/showthread.php?t=150965

virtuald, 2016-09-23:
Also related: Polymer.js provides a mechanism to make encapsulated HTML/JS data-driven components. It looks really cool and relatively simple (I think).

It's very similar to what @amorygalili was trying to do for the websim, but simpler. 

For a demo... https://github.com/googlecodelabs/polymer-first-elements/tree/master/icon-toggle-finished

closed 2018-01-20 by ErikBoesen