FRCDashboard/FRCDashboard #14: FRCDashboard not working with Electron in Windows 10

czhao39, 2017-01-18:
When I run `npm start` with the default code, a blank Electron window opens but no content appears. However, it does work in the browser with `python pynetworktables2js`. Also, all the content shows up properly in Electron in Linux (Fedora), so it seems like this issue is specific to Windows.

ErikBoesen, 2017-01-18:
Try pressing `Ctrl+R` to refresh. I've seen this bug around a couple times, and I'm guessing that's the problem.

czhao39, 2017-01-18:
That does not seem to do anything.

ErikBoesen, 2017-01-18:
Have you installed `pynetworktables2js`?

ErikBoesen, 2017-01-18:
If you have and the issue still persists, you can try running `start py -3 -m pynetworktables2js`.

czhao39, 2017-01-20:
Starting `pynetworktables2js` before running `npm start` worked. Thanks!

ErikBoesen, 2017-01-20:
I'm going to keep this open since this is an annoying workaround at best. Good luck this season regardless!

czhao39, 2017-01-22:
I'm sorry, I figured out the problem. I'm using a virtualenv, and I do not have the `py` command in the virtualenv. Therefore, `pynetworktables2js` was running in my host machine's python environment, which does not have all the necessary packages installed.

ErikBoesen, 2017-01-22:
Ah, got it. Good to hear. Good luck this season!

closed 2017-01-22 by ErikBoesen