FRCDashboard/FRCDashboard #16: What way do you guys go about retrieving your video feed from robot?

ThePfaffanater, 2017-02-04:
I mostly develop in Java and I have been wanting to create a nice dashboard for my robotics team. How do you guys go about retrieving the camera footage from your robot. I have found It quite difficult and the smart dash uses some very weird system for USB cameras. Do you guys use some sort of h.264 stream. I do not know much about networking unfortunately. Much thanks in advance.

ErikBoesen, 2017-02-04:
[As is detailed in the README, our team uses `mjpg-streamer`.](https://github.com/FRCDashboard/FRCDashboard#configuring-camera-feed)

ThePfaffanater, 2017-02-04:
does this require someone to be using Python with their robot development? How would it take an input?/multiple cameras

ErikBoesen, 2017-02-04:
I'm pretty sure the language of choice is irrelevant, so it should work with Java robot code as well. I'm not sure how to set up a multi-camera system, but you can search around and it should be pretty simple.

czhao39, 2017-02-11:
I'm using `cscore.MjpegServer` to listen on port 5801, which runs without error. However, I just get a Connection Refused error when I go to roborio-1123-frc.local:5801 or roborio-1123-frc.local:5801/?action=stream. Any ideas? (Not sure if I should ask this under the robotpy-cscore repo instead.)

ErikBoesen, 2017-02-11:
@czhao39 This isn't a problem with this dashboard, and I'm not hugely familiar with how `cscore` works. Probably a better place to ask would be on `robotpy-cscore`.

closed 2017-02-11 by ErikBoesen