FRCDashboard/FRCDashboard #18: The installer script for mjpg-streamer leads to a 404

StoneMoney, 2017-02-14:
This link in the readme: https://raw.githubusercontent.com/robotpy/robotpy-wpilib/master/installer/installer.py

auscompgeek, 2017-04-19:
It lives in its own PyPI package now. `pip install robotpy-installer`, then run `robotpy-installer`.

auscompgeek, 2017-04-19:
Either way, you don't actually need mjpg-streamer any more, WPILib's CameraServer class lets you have an MJPEG stream without using mjpg-streamer.

ErikBoesen, 2017-04-19:
Someone care to make a pull request clarifying this? I meant to get back to this and totally forgot.

ErikBoesen, 2018-01-17:
@ianzullo Look at this. In the README we shouldn't say that people should download `installer.py` as they can now use the package `robotpy-installer` or use WPILib's CameraServer class.

commit 07cf3f90 (2018-01-20): Modernize camera stream advice in README

Fix #18
closed 2018-01-20 by ErikBoesen