FRCDashboard/FRCDashboard #19: When I try to run pynetworktables2js, I get this error.

Jetlogan, 2017-02-15:
`Traceback (most recent call last):
  File "pyinstaller\shim.py", line 6, in <module>
  File "pynetworktables2js\__main__.py", line 81, in main
  File "pynetworktables2js\__main__.py", line 37, in init_networktables
AttributeError: Values instance has no attribute 'dashboard'
Failed to execute script shim`

Anyone know what this means?

czhao39, 2017-02-15:
Have you upgraded to the newest version of `pynetworktables2js`? That should fix the error.

Jetlogan, 2017-02-15:
I mean in the error it says pynetworktables2js so I guess.

czhao39, 2017-02-15:
But do you have the newest version? Run `pip install -U pynetworktables2js` to upgrade. This was an error in the previous release and should be fixed now.

Jetlogan, 2017-02-15:
I'm using the Windows version.

ErikBoesen, 2017-02-15:
In that case, `py -3 -m pip install -U pynetworktables2js`.

Jetlogan, 2017-02-15:
In CMD right?

ErikBoesen, 2017-02-15:
Yes.

ErikBoesen, 2017-02-24:
Did you figure this out?

closed 2017-02-24 by ErikBoesen