FRCDashboard/FRCDashboard #23: Cannot connect to robot

Bo887, 2017-06-20:
Upon attempting to connect to the robot (using the IP address of the roboRIO), I get this message:
[image]

I am using the default code provided with no modifications.

ErikBoesen, 2017-06-20:
@rakusan2 Any ideas? Looks like this is from your code.

rakusan2, 2017-06-20:
what version of wpilib-nt-client do you have?

Bo887, 2017-06-21:
1.2.6

AdamBusch, 2017-10-18:
I have this same issue as well, happening from a new project as well.

ErikBoesen, 2017-10-18:
@rakusan2 Any ideas?

rakusan2, 2017-10-18:
version 1.x is not able to deal with packets that have been split
version 2.x will while also giving a more detailed error report

rakusan2, 2017-10-18:
I should be finished with version 2 in two week since it will require an almost complete rewrite

rakusan2, 2017-11-22:
Sorry for the long delay but work kept getting in the way till I had forgotten.
On the other hand I had decided to hold out on 2.0 till I have more time and just make a quick fix on 1.x and release **1.3.0**
Based on the tests that I had done, this one is able to handle split packets.
I would appreciate a confirmation on that with your own code since I only have a hath running simulator on hand

closed 2018-01-17 by ErikBoesen