FRCDashboard/FRCDashboard #30: Problem with electron-builder and a possible alternative

rakusan2, 2017-11-27:
As I have been trying to upgrade the [electron-builder](/electron-userland/electron-builder) that this project uses I had discovered that the newest version requires Python during the package install, which will prevent people without Python from running a simple `npm install`. This project also can not stay on the current version that this project is using since since it does not support Electron v1.7.x

My solution would be to use [electron-packager](/electron-userland/electron-packager) which does not create an installation executable and rather sticks with a portable one. Another benefit is that from both Linux and Mac can generate each others files tho only Windows can generate the Windows files unless you use HomeBrew or Wine.
A con of this packager is that for Windows and Linux it creates a folder with libraries mixed with the executable.

This Packager can be used through [electron-forge](/electron-userland/electron-forge) which is able to create an installer, tho I have not yet been able to get it to cross compile on top of it being much slower than the packager alone

If anyone has a better plan to solve the issue of the builder or wants me to use the electron-forge then post a response else I will implement it during the next weekend

ErikBoesen, 2017-12-03:
Don't know why I haven't asked this yet, but why do we need a build system at all? I've always ran it from source, and I find that quite convenient.

ErikBoesen, 2017-12-03:
Also, who doesn't have Python?

rakusan2, 2017-12-03:
This will enable to copy the dist folder and put it on a system that does not have node, electron, python, or anything else other that the operating system and just run it without installing it
have a look at issue #12 

Also most people that I know from the robotics team that I have been part of do not have python and some do not even want to install it

ErikBoesen, 2017-12-03:
Don't only programmers need Python?

ErikBoesen, 2017-12-03:
I've always figured that non-technical users wouldn't be able to customize their dashboard anyway, thus it's not very important.

rakusan2, 2017-12-03:
The only reason that I have python is because my 3D printer and this project before my wpilib-nt-client needed it. Other than that my web, SQL, and C# programming do not need it.

rakusan2, 2017-12-03:
There is also the thing that I have not yet even considered learning it

ErikBoesen, 2017-12-03:
How would you not have Python installed?

ErikBoesen, 2017-12-03:
https://docs.python.org/3/using/windows.html
> Unlike most Unix systems and services, Windows does not include a system supported installation of Python.

Ahhh... okay.

closed 2018-01-09 by rakusan2