FRCDashboard/FRCDashboard #32: Problem with npm i

amstrudy, 2018-01-08:
I installed npm just fine, but when running npm i inside the FRCDashboard directory I get the following output:
[image]

What should I do? I'm running Ubuntu 16.04 on the 2013 MBP.

ErikBoesen, 2018-01-08:
You're using extremely old versions of Node and NPM. You have v4.2.6 and v3.5.2 respectively, whereas the newest versions available are 9.3.0 and 5.6.0. Update them, and you should be alright. Let me know if you have further issues.

closed 2018-01-08 by ErikBoesen

amstrudy, 2018-01-08:
How do I update? The latest versions via apt-get are what I have (v4.2.6).

ErikBoesen, 2018-01-08:
[This](https://askubuntu.com/questions/426750/how-can-i-update-my-nodejs-to-the-latest-version) might help you.

amstrudy, 2018-01-08:
When I try to run npm, I get this error:

`ERROR:` npm is known not to run on Node.js v4.2.6
`Node.js 4 is supported but the specific version you're running has
`a bug known to break npm. Please update to at least 4.7.0 to use this
`version of npm. You can find the latest release of Node.js at https://nodejs.org/

amstrudy, 2018-01-08:
Alright I fixed npm and tried restarting the FRCDashboard install:
[image]
Now I get an error when running npm run dist

ErikBoesen, 2018-01-08:
@rakusan2 This is your domain, I think.

rakusan2, 2018-01-08:
As I have mentioned in issue #30, the version of electron-builder that this project is currently using does not support electron 1.7.x.
Due to this, the fact that later versions require python, and that those versions have install problems on my system which has python, I had proposed pull request #31.

ErikBoesen, 2018-01-08:
@rakusan2 I totally forgot about merging that, I'm so sorry! I've merged it now, so this issue should be fixed. If for some reason it's not, you don't actually have to compile your dashboard at any point, you can just run it with `npm start` on your driverstation PC.

rakusan2, 2018-01-08:
Yup, 
Thanks

ErikBoesen, 2018-01-09:
@rakusan2 I don't remember if I've asked, but why do we need to be building the application into an executable in the first place? Is it not easier and more practical to simply run it from source...?

rakusan2, 2018-01-09:
Some people do not have the rights to install npm or just do not want to muck around with a terminal.
Also as I said earlier ask the person who had created issue #12 for more

ErikBoesen, 2018-01-09:
I've spoken with Dustin about this. If they don't want to install `npm` or muck around with terminal, why are they using a framework which requires doing that to be useful?

rakusan2, 2018-01-09:
Do you want to lock this dashboard to only developers or can the developer customize the dashboard, package it and then give it to the users?
If the later than how proficient do you think that those users with a computer terminal?

rakusan2, 2018-01-09:
Also the driver station can be on a school computer where only the teachers have install privileges.

ErikBoesen, 2018-01-09:
All a team needs is one developer competent enough to make the desired changes, then clone their code to one driverstation computer. In what situation would anyone be distributing the software to users?

ErikBoesen, 2018-01-09:
Your school won't give you admin permission on a driverstation computer? That's pretty harsh.

It is a valid justification for keeping building support, though, I think. I'd like to simplify some things (like the README), though.

ErikBoesen, 2018-01-09:
Hey, try this when you get a chance: https://stackoverflow.com/questions/37029089/how-to-install-nodejs-lts-on-windows-as-a-local-user-without-admin-rights

rakusan2, 2018-01-09:
For example I might want to use this project to make a customized dashboard for my team who do not know js or html and then they might want to send it of to other teams who might also not know them.
This is very common here in Canada

rakusan2, 2018-01-09:
I know this install process but it is still too complex by what I know of my team

ErikBoesen, 2018-01-09:
Are all those teams using the same camera hosts/ports and NetworkTables variable names, and unable to install Node on one computer?

As a side note, couldn't teams without any programming capability just use SmartDashboard? I conceived this project as a tool for teams who wanted deep customization of their control system. If a team isn't actually changing anything I think they might be better served with more traditional driverstation dashboard apps.

ErikBoesen, 2018-01-09:
Can't you get access to the driverstation computer to install node?

rakusan2, 2018-01-09:
I am currently going to school that is 800km away so that is going to be difficult

ErikBoesen, 2018-01-09:
Could you explain the situation in a bit more detail?

rakusan2, 2018-01-09:
I mentor my old team by reviewing the code that they have pushed to github and communicate with them through slack. Due to this I know how the dashboard is going to be used and also that team contains programmers that know just a little of C#, little less of Java, and hardly anything else.

I personally wanted to make them a dashboard that they can show of to the other teams tho it is most likely not going to be used unless it works right from the start doe to the head mentor not liking delays and some of the drivers not being very proficient with computers.

Also I am trying to make the use of it as easy as possible for non programmers (the drivers and their mentors) which means that if I give them a terminal, they will most likely ask for an executable anyways making it easier for me to just send them a package and maybe a link to the code.

Overall my main reason is that the code is useless to the ones running the dashboard so why bother giving it to them and teaching them how to use it

ErikBoesen, 2018-01-09:
Ah, so you're an alumnus, that makes more sense.

Though writing code as a mentor is a vague violation of FIRST rules, and this situation is a bit of an edge case, I think it's reasonable that we keep support for building. I've update the README to make it clear that this step is optional, though.