FRCDashboard/FRCDashboard #37: Network Tables

Bananajojo123, 2018-01-20:
How do we grab network table values and post network table values?

ErikBoesen, 2018-01-20:
For fetching values from NetworkTables, you can use `NetworkTables.getValue('/path/to/value')`. For putting them, you can use `NetworkTables.putValue('/path/to/value', value)`.

You can also use
```js
NetworkTables.addKeyListener('/SmartDashboard/example_variable', (key, value) => {
    // Do things with value
}
```
to create a listener for the value being changed.

closed 2018-01-20 by ErikBoesen

Bananajojo123, 2018-01-20:
What is the name of the network table? Can you set the name of the network table?

ErikBoesen, 2018-01-20:
You will need to give the full path to the variable. So, 'SmartDashboard/valuename` if you're using the `SmartDashboard` table, or `robot/valuename`, etc.

Bananajojo123, 2018-01-20:
What libraries do I have to import?

ErikBoesen, 2018-01-20:
None.

Bananajojo123, 2018-01-20:
In the console, IPC is not defined in the NetworkTables.js file. In the ui.js file, require is not defined.

ErikBoesen, 2018-01-20:
Yup, I just noticed some issues with our implementation. In theory you should not have to do any imports yourself. We will be fixing this issue pretty soon. See #40 for some more information.

ErikBoesen, 2018-01-20:
@rakusan2 Do you have any idea what could be going wrong? This seems like a pretty integral issue.

rakusan2, 2018-01-20:
Sorry, back when you removed tryIPC.js and declared the ipc variable somewhere else, I forgot to inform you that based on index.html, networktables.js runs before ui.js.
Therefor the ipc declaration has to be at the beginning of networktables.js

Bananajojo123, 2018-01-20:
Thanks for that! But what about require is not defined in all of the .js files

ErikBoesen, 2018-01-20:
Working on fixing this, I'm also fixing #38 in the process. Will let you know when I've finished.

rakusan2, 2018-01-20:
After moving the ipc declaration to networktables.js everything seems to be working fine for me including require. 
So I have no idea what is causing require to not be defined for you

Bananajojo123, 2018-01-20:
It's only relevant when you use the network tables editor in the dashboard. It's one of the components.

ErikBoesen, 2018-01-20:
Alright, this should be fixed; feel free to pull changes.

ErikBoesen, 2018-01-20:
@Bananajojo123 I recommend using OutlineViewer, as the Tuning interface has been [deprecated](#39). (Note that you will still be able to use it through its [addon](https://github.com/FRCDashboard/addon-Tuning) if you wish.)

Bananajojo123, 2018-01-20:
Because you added connection.js, it has require is not defined. The networktables.js also has the same error. And in ui.js, Cannot read property 'addKeyListener' of undefined at ui.js:44.

ErikBoesen, 2018-01-20:
That's really odd. This worked fine for me, but shouldn't have. I'll fix this momentarily.

Bananajojo123, 2018-01-20:
Would you like my code?

Bananajojo123, 2018-01-20:
No, only the error that was being caused from connection.js and not networktables.js.

ErikBoesen, 2018-01-20:
If you want to share your repository with me, feel free.

rakusan2, 2018-01-20:
Are you running this through electron? and what version?

Bananajojo123, 2018-01-20:
Here: https://github.com/MRT3216/MRT3216-2018-PowerUp/tree/master/ROBOTICS/FRCDashboard-master/FRCDashboard-master

rakusan2, 2018-01-20:
After moving the ipc declaration to networktables.js I ran your code and got this 
[image]

rakusan2, 2018-01-20:
I would recommend updating it with the latest changes to this project

Bananajojo123, 2018-01-20:
done

Bananajojo123, 2018-01-20:
try again

rakusan2, 2018-01-20:
The script connections.js should run after networktables.js but you have that script in the head

rakusan2, 2018-01-20:
You are also missing an element with `id="arm-position"`

Bananajojo123, 2018-01-20:
done

Bananajojo123, 2018-01-20:
It's still giving me the error

ErikBoesen, 2018-01-20:
Putting `<script>` tags in the `<head>` will cause them to be run before the page loads. Typically it's a good idea to put them (in order of dependency) immediately before the closing `</body>` tag.

Bananajojo123, 2018-01-20:
I did move them right before the </body>

Bananajojo123, 2018-01-20:
I put them in order, near the </body>

ErikBoesen, 2018-01-20:
By the way, I'd recommend that you think about moving your robot code, dashboard code, etc. into separate repositories. This allows them to be managed much more easily.

rakusan2, 2018-01-20:
After the change that you made, the only error that I see now is in ui.js and it is caused by no element having an id of "arm-position"

ErikBoesen, 2018-01-20:
Also, typically it's not a good idea to download a zip of a repository, at least when you're planning to put it into a repository. Don't ever commit zip files. The Tuning addon shouldn't have its full directory present, either.

ErikBoesen, 2018-01-20:
(This isn't so much related to this issue, just some general tips on your code.)

rakusan2, 2018-01-20:
If I may add my own tip.
Add bin folders to your .gitignore file since those folders are usually large and need to be recreated on every system anyway

Bananajojo123, 2018-01-20:
So I made some changes and the new repository is here: https://github.com/MRT3216/Dashboard.
I still can't figure out what's going on. 
[image]

rakusan2, 2018-01-20:
This need to run through electron not through a browser as that functionality had been removed

ErikBoesen, 2018-01-20:
Oh, yeah, that's the problem. Glad we could help! Let us know if you need anything.

ErikBoesen, 2018-01-20:
I see you've moved to a new repository, good choice.

Some other recommendations for your code:
* Commit titles should be written in the imperative tense. So, 'Add...' rather than 'Added...'. They should additionally be descriptive (avoid "more updates", etc.) and use proper capitalization.
* Storing binary-format documents in git is typically a bad idea unless you actually have to. Rather than a Word document, might I suggest you use a Markdown file? Markdown can easily be viewed both natively on GitHub or as plain text, and doesn't require that users have a closed-source product installed, nor that they spend time opening such a product. You could recreate your current document by creating `control_system.md` with the following content:
```md
* High/low gear drive train
* Lifter (button)
* Intake (in and out)
* Climber (winch) (up and down)
* Tank drive
    * Click one joystick and push other joystick to drive straight
```
* Your git config seems to be broken; you can enter the following commands to help GitHub properly associate your commits with your user account:

Bananajojo123, 2018-01-20:
Ok, thanks! 👍

Bananajojo123, 2018-01-20:
Also I saw that you made even more changes to the code. I was wondering if I should add that into my code.

ErikBoesen, 2018-01-20:
Feel free. None of it is major, but you might benefit.

Just so you know, one of the reasons to keep your code in a separate repository is that you can use git to pull changes to the code and git will help you automatically integrate updates (and resolve conflicts if necessary).

Bananajojo123, 2018-01-22:
Also, Is there any way that you can launch developer options in the application?

ErikBoesen, 2018-01-22:
@Bananajojo123 Comment out this line in `src/main.js`:
```js
mainWindow.setMenu(null);
```
You'll then be able to access the menus to open the developer options.

rakusan2, 2018-01-22:
The second argument to `addKeyListener()` has to be a function.
This argument is currently not being checked but I will soon make a pull request where it will be