FRCDashboard/FRCDashboard #40: Electron app won't connect to localhost

Chandi-95, 2018-01-20:
We've set up everything correctly on our Driver Station, but the application is stuck on connecting to the localhost. We haven't touched anything in the code either, except for changing the timer. Is there a way to bypass the connection to localhost/fix the connection issue?

[image]

ErikBoesen, 2018-01-20:
Are you hosting a simulator on your computer? Otherwise, there won't be anything on `localhost` to connect to.

ErikBoesen, 2018-01-20:
Hold on, you're right, there seems to be an issue with some recent changes. Give me a moment, I'll see what I can do.

ErikBoesen, 2018-01-20:
This should be fixed now. Feel free to pull changes.

closed 2018-01-20 by ErikBoesen