FRCDashboard/FRCDashboard #50: The Dashboard doesn't connect to 2018 Robot Code

eAUE, 2018-02-01:
The FRC Dashboard cannot connect to 2018 Robot code. No errors are thrown, the dashboard just displays the "Connecting..." button all the time. IP address is correct and it still works with 2017 robot code.

ErikBoesen, 2018-02-01:
Gonna need some more details on this. Can you show us a log (comment out `mainWindow.setMenu(null);` in `main.js` if you need to show the menu to open the developer tools)?

ErikBoesen, 2018-02-01:
Just for reference, I'm using it on my 2018 robot code right now and everything's just fine.

eAUE, 2018-02-01:
Took Screenshot of console. Only code modifications I made was commenting out the `mainWindow.setMenu(null);`. 
[image]

ErikBoesen, 2018-02-01:
Just to check:

1. You are connected to the robot's network
2. You've inputted the correct hostname into the box

Correct?

eAUE, 2018-02-01:
Yes. The DriverStation can connect to the robot, but the Dashboard doesn't connect.

ErikBoesen, 2018-02-01:
What hostname are you giving it?

eAUE, 2018-02-01:
We were using the IP address of the Roborio because that's what's worked the best. We tried using roborio-xxxx.local with our team number filled in as well as roborio-xxxx-FRC.local.

ErikBoesen, 2018-02-01:
Okay, hmm. @rakusan2, do you have any ideas?

eAUE, 2018-02-01:
I'm wondering if the robot is somehow misconfigured. Is this possible?

ErikBoesen, 2018-02-01:
Yeah, that is possible.

rakusan2, 2018-02-01:
try roboRIO-####-FRC.local with the capital RIO

ErikBoesen, 2018-02-01:
Have you figured this out?

eAUE, 2018-02-02:
I've tried a couple different things. I've got it working on another robot with the 2018 firmware so I'm going to troubleshoot it more on the problem robot. Thanks for your help with this issue.

ErikBoesen, 2018-02-02:
Happy to help! Feel free to inquire if you've got any further questions!

closed 2018-02-02 by ErikBoesen

eAUE, 2018-02-03:
@ErikBoesen I'm actually still having trouble with connecting at all again. We figured out that the robot we got it to connect to was actually running 2017 code, not the 2018 code that we thought it was running. 

I was able to connect using just running a networktables debug python script (see below) so I think the issue is the bridge between networktables and the dashboard. Any thoughts about how my pc might be misconfigured to allow this to happen?

[image]

```python
#!/usr/bin/env python3
#
# This is a NetworkTables client (eg, the DriverStation/coprocessor side).
# You need to tell it the IP address of the NetworkTables server (the
# robot or simulator).
#
# When running, this will continue incrementing the value 'dsTime', and the
# value should be visible to other networktables clients and the robot.
#

import sys
import time
from networktables import NetworkTables

# To see messages from networktables, you must setup logging
import logging
logging.basicConfig(level=logging.DEBUG)

if len(sys.argv) != 2:
    print("Error: specify an IP to connect to!")
    exit(0)

ip = sys.argv[1]

NetworkTables.initialize(server=ip)

sd = NetworkTables.getTable("SmartDashboard")

i = 0
while True:
    print('robotTime:', sd.getNumber('robotTime', 'N/A'))
    
    sd.putNumber('dsTime', i)
    time.sleep(1)
    i += 1

```

ErikBoesen, 2018-02-03:
@rakusan2

eAUE, 2018-02-03:
Update on this: A couple times I was able to make my version of the Dashboard connect to the robot just after doing `npm update electron -s` in the command line in the directory of the project, however this would only work the first time after doing the command.  I would have to call `npm update electron -s` each time I wanted to launch the program.

eAUE, 2018-02-03:
So I updated it, ran it and it worked the first time but not after that. 

[image]

rakusan2, 2018-02-03:
Can I ask you to run the NTClients test on the server and send me the output?
[Here](https://github.com/rakusan2/FRC-NT-Client/releases/download/v1.5.1/NTClient.zip) is the transpiled version

eAUE, 2018-02-03:
What do you mean by running it on the server?

rakusan2, 2018-02-03:
I mean connect the test script to the server
so that I can determine if the problem lies with the client or the dashboard

eAUE, 2018-02-04:
I ran the test.js located in the NTClient\FRC-NT-Client\test folder. It came out with this error when I tried to connect. 
```
[Folder Path]\NTClient\FRC-NT-Client\test\test.js:11
        throw err;
        ^
Error: connect ECONNREFUSED 127.0.0.1:1735
    at Object._errnoException (util.js:1024:11)
    at _exceptionWithHostPort (util.js:1046:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1182:14)
```

rakusan2, 2018-02-04:
Thank you,
But it seams  that I have made an error in the file that I have linked resulting in it only being able to connect to localhost
Sorry for that.

Do you mind trying it again with [this new one](https://github.com/rakusan2/FRC-NT-Client/releases/download/v1.5.2/FRC-NT-Client.zip)?
In this one I have made it so that you can call `npm test [address of the server]` so that you do not have to edit the code.
And don't worry this time I had tested it with my server.

eAUE, 2018-02-06:
I ran the program for a couple of seconds and it gave a lot of output. Seems to have been able to connect to the robot.  

Copied the output into a text file and attached it. 

[DashboardDebug.txt](https://github.com/FRCDashboard/FRCDashboard/files/1697260/DashboardDebug.txt)

eAUE, 2018-02-06:
I also keep getting this error occassionally:

rakusan2, 2018-02-06:
That huge amount of output is there since the test spits out everything that it gets without a filter.
From which I can tell that that during that run everything worked as it was meant to.

rakusan2, 2018-02-06:
Can you tell me the version of wpilib-nt-client in that dashboard?

rakusan2, 2018-02-06:
That version has that [bug](https://github.com/rakusan2/FRC-NT-Client/issues/5).
It has been fixed in 1.3.0 and I would recommend upgrading to the latest 1.5.2.

eAUE, 2018-02-06:
How do I update it then? `npm update` returns an error:

```
D:\Programming\2706 New Dashboard\Dashboard\2706Dashboard>npm update
npm ERR! code EINVAL
npm ERR! EINVAL: invalid argument, read

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\kylek\AppData\Roaming\npm-cache\_logs\2018-02-06T01_04_12_422Z-debug.log

D:\Programming\2706 New Dashboard\Dashboard\2706Dashboard>
```

eAUE, 2018-02-06:
I updated it, connectivity's working now. Not sure why `npm update` didn't update it...

rakusan2, 2018-02-06:
Great.
My thought on that problem would be try to stay on the latest LTS version of node unless you want to try new features and if you installed a different version of npm that what came with node than try updating it.