FRCDashboard/FRCDashboard #51: & missing in main.js line 56

Bananajojo123, 2018-02-01:
You have to add a & in line 56 of main.js. The line looks like this client.Assign(mesg.val, mesg.key,(mesg.flags & 1) === 1);. Add another & near the current one.

ErikBoesen, 2018-02-01:
This is a bitwise AND operation. It's supposed to be this way. [More information](https://en.wikipedia.org/wiki/Bitwise_operations_in_C)

closed 2018-02-01 by ErikBoesen

Bananajojo123, 2018-02-01:
It works with &&

ErikBoesen, 2018-02-01:
No, it doesn't. `&&` and `&` are fundamentally different operations. While it may work the same in some cases, you're doing a different operation and the outputs will be different. `x & 1` is `true` if the last bit of `x` is `1`. `x && 1` is `true` if `x` is anything other than 0. We use the bitwise AND here intentionally, it does not need to and should not be changed.