FRCDashboard/FRCDashboard #60: NetworkTables cannot access values

eAUE, 2018-02-10:
All the values in the `NetworkTables` object are null. `NetworkTables.isRobotConnected()` returns `true` but if we try to get any values from the robot, they return undefined. `pynetworktables2js` is installed, and all dependencies are up to date. Running `NetworkTables.getKeys()` returns an empty array. I can confirm using the FRC outline viewer that the keys exist and are accessible. 

[NetworkTablesNotWorkingTrim.zip](https://github.com/FRCDashboard/FRCDashboard/files/1713588/NetworkTablesNotWorkingTrim.zip)

ErikBoesen, 2018-02-10:
@rakusan2

rakusan2, 2018-02-11:
Sorry, 
This has happened due to a bad optimisation

closed 2018-02-11 by ErikBoesen

eAUE, 2018-02-13:
This still doesn't work.

rakusan2, 2018-02-13:
Have you updated the wpilib-nt-client to?

eAUE, 2018-02-13:
It's at 5.6.0 and I pulled the changes made to the repository.

[image]

rakusan2, 2018-02-13:
The highest version that I have published is 1.6.0

ErikBoesen, 2018-02-13:
@eAUE That command is getting the version of `npm`, not `wpilib-nt-client`.

rakusan2, 2018-02-13:
Try the command `npm ls - - depth=0`

eAUE, 2018-02-13:
Oh yeah, sorry. Here's the real screenshot:
[image]

rakusan2, 2018-02-13:
In main.js does the `client.addListener` have a second parameter with a value of `true`?

eAUE, 2018-02-13:
Yes, it does: 
[image]

rakusan2, 2018-02-13:
Can you add this line to main.js `client.startDebug('NT-Client', 1)` and send me the output from the console?

eAUE, 2018-02-13:
Yeah, I'll be able to test that on Wednesday at the earliest.

commit e071c350 (2018-02-11): Move addListener to ready event

Fix #60