FRCDashboard/FRCDashboard #65: Feature request: Automatically try to reconnect after unexpected disconnect

Chris2fourlaw, 2018-02-11:
Helpful when robot code restarts.

rakusan2, 2018-02-12:
I will need to update the client to ensure that all of the keys sync once it connects instead of the server just sending what it has.
I should be done with that tonight.

rakusan2, 2018-02-16:
Sorry that it took longer than I thought but due to my busy week I have been making lots of bugs that needed to be fixed.

closed 2018-02-18 by ErikBoesen

Chris2fourlaw, 2018-02-19:
2f58b5d created a problem described in #70. Please reopen!

ErikBoesen, 2018-02-19:
This is a separate issue.

Chris2fourlaw, 2018-02-20:
No, the `client.setReconnectDelay(1000);` causes this

ErikBoesen, 2018-02-20:
What I mean is that we have implemented the feature requested by this issue, and #70 is open to address the bug in the implementation.

ErikBoesen, 2018-02-20:
@rakusan2 Can you fix this? And do you mind testing more stringently in the future?

rakusan2, 2018-02-20:
I have fixed this in wpilib-nt-client@1.7.1
And I will be getting into the habit of trying everything in the simulator before creating a pull request

ErikBoesen, 2018-02-20:
Awesome, thanks Thomas! @rakusan2