FRCDashboard/FRCDashboard #78: Dashboard doesn't connect to robot with 2019 firmware

Nicholas264, 2019-01-12:
The latest commit of FRCDashboard is incompatible with either the router firmware or the roborio firmware (most likely this) updated in 2019. Neither the 10.xx.xx.1 address or roborio-xxxx-frc.local connects to the robot when replaced with our team number when it is on the WiFi network. We used [this](https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/144986-programming-your-radio) to configure our radio to the current firmware. This caused it to stop connecting. It does work, however, with a robot that has firmware from Recycle Rush in 2015. I am using commit 6481dbb with nothing changed but the `address.value` line in `connections.js` to our team number. Has the address value changed for connecting now, or is it another issue?

ErikBoesen, 2019-01-12:
Can you connect to those addresses at all, outside of FRC Dashboard? This doesn't seem like it'd be an FRC Dashboard problem since the software exclusively communicates by exchanging NetworkTables values.

Nicholas264, 2019-01-12:
I got it to work! I changed this code in `main.js` to get the dashboard to autoconnect to the correct address. The roborio-xxxx-FRC.local wasnt working for us, but the `10.47.38.2:1735` ip address and port did allow network tables to exchange data. The function `setLogin` was unneeded for connecting in `connection.js`.
```javascript
//main.js
ipc.on('connect', (ev, address, port) => {
	address = "10.47.38.2"; //default would be 10.xx.xx.2
	port = 1735; //port is the one specified here for network tables https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/144986-programming-your-radio#programmed_configuration
	console.log(`Trying to connect to ${address}:${port}`);
	let callback = (connected, err) => {
		console.log('Sending status');
		mainWindow.webContents.send('connected', connected);
	};
	client.start(callback, address, port);
});
```

closed 2019-01-13 by Nicholas264

samperlmutter, 2019-03-14:
My team is having this same problem, but the solution in this thread didn't seem to fix it. We're using the latest rio firmware version, and our code is [here](https://github.com/Yeti-Robotics/yeti-dashboard). We're also able to connect via a local networktables client.

ErikBoesen, 2019-03-14:
@rakusan2 Any ideas?

rakusan2, 2019-03-14:
@samperlmutter Have you tried the address 10.35.6.2
Also, are you getting any errors in the console?

samperlmutter, 2019-03-14:
Yes we've tried ipv4 and mdns with and without specifying the port. The chrome console is reporting that it connected, but the terminal keeps saying `Sending status`

Nicholas264, 2019-03-14:
@samperlmutter I've had that repeated message appear in logs while the dashboard was connected and successfully sending and receiving data. Despite it logging that it was connected, sending test keys/values could verify if it actually is connected or not. If you haven't tried sending test values yet, try modifying robot code to send an example value to the dashboard.

samperlmutter, 2019-03-14:
I'm not sure what you mean by test keys or how they would be different from real keys. The real keys we're sending from the robot code don't trigger the network tables listeners on the dashboard. I've put logging in the listeners and they're not being fired. If I try and use it connected to network tables running locally on my laptop it works perfectly.

Nicholas264, 2019-03-14:
By "test keys" I meant keys to test if it could successfully send data. However, it sounds like it you have already done that. How I originally solved the connection issue was by using wireshark to log local network requests on the wifi network. I could see the IP and port that data was being sent over while correctly connected to network tables outside of the dashboard. Since you can recieve and send values over network tables, that solution might work for you.

samperlmutter, 2019-03-14:
I'll try that later today, thanks.

samperlmutter, 2019-03-14:
Just finished looking at wireshark and it's not showing me anything I didn't expect. 
`roborio-3506-frc.local (10.35.6.2)` with destination port 1735