FRCDashboard/FRCDashboard #85: No documentation for client-side NetworkTables interface

cjdenio, 2020-02-12:
First off, there may be documentation that I'm missing. Please disregard this issue if so :smile:

@ErikBoesen, what are your thoughts on using [JSDoc](https://jsdoc.app) to generate documentation for the client-side `NetworkTables` interface? There's a wealth of helpful functions in there, but it's hard to find out that they exist. :wink: 
You've already done the hard work of putting JSDoc comment blocks in the code, so it would be pretty easy to do.

Here's a preview of what it might look like:
[image]

ErikBoesen, 2020-02-12:
First of all, I looked through your work a little bit, and wow. Incredible for your age. Your future is bright, keep it up man. :)

Second of all, sure, if you're willing to take this on I'm fine with it! I'm a little too busy with college classes to handle it (plus I'm not on an FRC team anymore), but you're welcome to contribute. Thanks for taking the time!

cjdenio, 2020-02-12:
Thank you! I wouldn't call myself _that_ incredible, but I appreciate the encouragement! :smile: 

Sorry, I should have clarified. I'm not asking you to write the documentation; I'll gladly do it! I can completely understand that you're busy :wink: 

I threw up a PR, #86 , that fixes some errors with the JSDoc tags.

Thank you for building this amazing software with Electron; it makes it super easy for me to modify the core (and contribute to it) :electron: