FRCDashboard/FRCDashboard #87: Networktables connection stuck

Repulser, 2020-02-23:
Hey, we're experiencing an issue where once we hit connect it gets stuck on "Trying to connect" in the console and the callback never gets called.
Already tried using direct ip, mDNS, etx Anyone have any idea what's going on?

Love the project btw!

Repulser, 2020-02-23:
Literally a second after posting I found the issue, if you wait 2-3 seconds before hitting connect it works since the IPC is already connected, but if you do so before the dash won't connect.

Repulser, 2020-02-23:
It seems to work sometimes and sometimes it hangs.

ErikBoesen, 2020-02-23:
@AndrewLester @rakusan2

AndrewLester, 2020-02-24:
We are having some issues with the NT Client calling our robot connection listener with a disconnect notification despite our cameras being loaded and NT being enabled. It may be related to this, so we're going to do some debugging today.

AndrewLester, 2020-02-24:
@Repulser one question, are you running your dashboard on windows or MacOS. Most of our problems occur on windows.

ErikBoesen, 2020-02-25:
Would assume Windows, unless the driverstation has been ported to MacOS since I graduated.

cjdenio, 2020-02-25:
Sadly, the Driver Station is still Windows-only.

ErikBoesen, 2020-02-25:
😢

Repulser, 2020-02-25:
The dashboard does seem to also run won MacOS, and it happens on both platforms.

AndrewLester, 2020-02-25:
Do you know if your network tables server on the robot is running on the default port?

Repulser, 2020-02-25:
@AndrewLester it is

rakusan2, 2020-02-25:
Try starting the debugger with `wpilib_NT.startDebug('NT', 2)`
This will console log status (0), message types (1), and messages (2) under the name NT
Other than that sorry but I currently can only test my code on a simulator which does not show any problems.

AndrewLester, 2020-02-26:
@Repulser try removing the client.start() call which tries to connect to localhost. (in main.js with the comment above it saying "attempts to connect to localhost"). That fixed our windows connection problem.

AndrewLester, 2020-02-26:
Note that deleting that line will prevent you from connecting to the simulator.

CrispyBacon1999, 2020-02-27:
My team's getting the same error. We tried removing the `client.start()` but it didn't change anything.

CrispyBacon1999, 2020-02-28:
Can't connect on both windows and Linux.

CrispyBacon1999, 2020-02-28:
After a bunch of testing, attempting to connect anywhere from 3-10 times makes it work, not really sure what that is, but it does actually attempt a connection and succeeds after several attempts.

AndrewLester, 2020-02-28:
@Repulser  @CrispyBacon1999  Try checking our [main.js](https://github.com/frc1418/2020-dashboard/blob/master/src/main.js) file for this season. It connects to the robot on the first try with macos and windows. We've made a few changes from the main.js file in FRCDashboard, so I'll probably do a git diff and find out what we may have fixed unintentionally at some point. It also doesn't use the connect.js file to initiate the connection.

Repulser, 2020-02-28:
Since our competition was yesterday I quickly ported things over to the pynetworktables2js framework which was rather a breeze. But I will definitely look into this for the next competition and let you know. Thanks!

ErikBoesen, 2020-02-28:
FRCDashboard used to use `pynetworktables2js` for connection, frankly I've never been a fan of the Node-based connection system so I'd be supportive of migrating back to that platform and simplifying the connection logic a whole lot. Up to you guys though.

Repulser, 2020-02-28:
That explains why porting it over was so simple, the API seems to have stayed between the two versions, pynetworktables2js has proven very stable for us in a competition environment, I support moving back to it. This means it'll be much easier to maintain over the years and can focus more on Dashboard functionality.

ErikBoesen, 2020-02-29:
@AndrewLester  can you work with @Repulser to migrate back to pynetworktables2js?

JubiusII, 2020-03-01:
I've been having the same problem for the past 2 days, so I gave in and started from scratch (I didn't do much so I'm not too cut up about it). I found out that this problem always happens when I delete the example value from index.html. I decided to keep and just hide it behind some stuff. works fine now.

ErikBoesen, 2020-03-01:
@AndrewLester ^

CrispyBacon1999, 2020-03-02:
What's the process look like to switch back to pynetworktables? My team's debating between that or trying to get a fix going with the nodejs version.

AndrewLester, 2020-03-12:
How about a new branch is made for the pynetworktables2js version, but the main build still uses wpilib-nt-client. I feel like the NodeJS implementation just fits in nicely with the electron app code, should it work properly. Since it _does_ work on occasion, I don't think we should just remove it.

ErikBoesen, 2020-03-12:
Up to you on the implementation, just get it done soon if you can. Although I don't know how much it will actually matter given the status of certain competitions...