FRCDashboard/FRCDashboard #89: Error running FRCDasshboard

DumboWumbo, 2020-03-10:
I downloaded the files, then ran npm install in the directory. However after running npm start it gave me this list of errors.

> FRCDashboard@2.1.0 start C:\Users\student\AppData\Local\Packages\Microsoft.MicrosoftEdge_8wekyb3d8bbwe\TempState\Downloads\FRCDashboard-master (1)\FRCDashboard-master
> electron .

npm ERR! code ELIFECYCLE
npm ERR! errno 3221225781
npm ERR! FRCDashboard@2.1.0 start: `electron .`
npm ERR! Exit status 3221225781
npm ERR!
npm ERR! Failed at the FRCDashboard@2.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

I would appreciate any help!!!

cjdenio, 2020-03-10:
Is this all you're seeing? It seems to indicate an unknown issue spawning the Node process.

Without further information, my best recommendation would be to copy the `FRCDashboard-master` folder into a non-temp directory, like `C:\Users\student`

ErikBoesen, 2020-03-10:
One thing that might help you file-manage better would be to actually clone the repository rather than just downloading the zipped files. https://git-scm.com/book/en/v2/Git-Basics-Getting-a-Git-Repository